utexas-bwi/segbot #30: segbot: clean up branches in this repository, delete most of them

jack-oquin, 2015-03-06:
There are currently 10:

- **master**
- active:
  - **segbot2**: almost ready to merge
  - **new_lazer**: already merged into segbot2, delete now?
  - **piyushk/new_robot**: work in progress?
- stale:
  - **devel**: this version was released to ROS Groovy. It's no longer in use, but references exist in ``rosdistro/groovy/distribution.yaml`` and hence the ROS wiki.
  - **navigation_testing**: has one small change that was never merged into master
  - **cleanup_segbot_description**, **stable**, **add_band_repair**, and **robot_state_publisher**: all have been merged into master. None has been modified in the last year or two.

jack-oquin, 2015-03-06:
Barring objections, I recommend immediate deletion of these: 
- cleanup_segbot_description
- stable
- add_band_repair
- robot_state_publisher
- new_lazer

@jsinapov: please decide what to do with **navigation_testing**

I will ask on *ros-release* how to drop support for Groovy and the **devel** branch.

Let's plan to delete **segbot2** when #31 is finally merged.

piyushk, 2015-03-06:
@jack-oquin I agree with your recommendation. I will delete the branches later today.

jack-oquin, 2015-03-06:
On reflection, I recommend leaving **devel** alone. It's the easiest solution.

jack-oquin, 2015-03-09:
I deleted most of the stale and already-merged branches:

- cleanup_segbot_description
- stable
- add_band_repair
- robot_state_publisher
- new_lazer

They can be restored if somebody thinks it was a mistake. :smile:

jack-oquin, 2015-03-16:
@jsinapov: I am leaving **navigation_testing** alone, but please delete if you no longer need it.

closed 2015-03-16 by jack-oquin