utexas-bwi/segbot #35: segbot_sensors: add diagnostics and arduino to launch scripts

jack-oquin, 2015-04-04:


jack-oquin, 2015-04-04:
@maxsvetlik: this stripped-down arduino.launch should be ready to invoke from some appropriate place in the segbot_v2.launch files.

jack-oquin, 2015-04-04:
I think this is fixed, but will leave it open until one of us tests it on the real robot.

piyushk, 2015-04-04:
There seem to be some failing tests because of missing manifest dependencies:
```
-- run_tests.py: verify result "/home/rosbuild/hudson/workspace/devel-hydro-segbot/ARCH_PARAM/amd64/UBUNTU_PARAM/precise/label/devel/test_results/repos/segbot_bringup/roslaunch-check_launch_segbot_v1.launch.xml"
Built target _run_tests_segbot_bringup_roslaunch-check_launch_segbot_v1.launch
Scanning dependencies of target _run_tests_segbot_bringup_roslaunch-check_launch_segbot_v2.launch
FAILURE:
[/tmp/test_repositories/src_repository/segbot/segbot_bringup/launch/segbot_v2.launch]:
	Missing manifest dependencies: segbot_sensors/manifest.xml: rqt_robot_monitor, diagnostic_aggregator
cannot find package [diagnostic_aggregator] for node [aggregator_node]
cannot find package [rqt_robot_monitor] for node [rqt_robot_monitor]
unable to find node [diagnostic_aggregator/aggregator_node]: diagnostic_aggregator
```

http://jenkins.ros.org/job/devel-hydro-segbot/ARCH_PARAM=amd64,UBUNTU_PARAM=precise,label=devel/24/console

jack-oquin, 2015-04-05:
``segbot_sensors`` had no ``roslaunch_add_fiile_check()`` for the diagnostics launch file.

I will add that test and the missing dependencies.

closed 2015-04-15 by jack-oquin