utexas-bwi/segbot #43: segbot_sensors: NanToInfFilter laser filter missing on segbot v1

jack-oquin, 2015-05-20:
Starting **marvin** on Hydro using the following command:
```
roslaunch segbot_bringup segbot_v1.launch --screen 
```
These errors appear:
```
process[kinect_laser_filters-16]: started with pid [26408]
[ERROR] [1432155745.443302262]: Couldn't find filter of type segbot_sensors/NanToInfFilter
process[hokuyo_laser_filters-17]: started with pid [26521] 
[ERROR] [1432155745.538516972]: Couldn't find filter of type segbot_sensors/NanToInfFilter
```

I saw navigation failures that may have been caused by failure to clear a previously-observed moving obstacle. Unclear yet whether that is related to the messages above.

jack-oquin, 2015-05-20:
It looks like that filter *should* be loaded for the segbot v1 machines.

piyushk, 2015-05-26:
@jack-oquin That's correct. Let's see if we can reproduce this error tomorrow. I think the workspace on marvin might be messed up.

jack-oquin, 2015-05-26:
Yes, I'll see you in the morning.

Pretty sure I created a new workspace before running any tests last week.

jack-oquin, 2015-05-27:
There was some problem on **marvin** with resolving filter plugins. Piyush did an apt-get upgrade for a large number of down-level packages on that laptop, which seems to have eliminated the problem.

closed 2015-05-27 by jack-oquin