utexas-bwi/segbot #50: Multiple packages found with the same name

shiqizhang6, 2015-07-30:
Multiple packages found with the same name "segbot_gazebo":
- segbot/segbot_gazebo
- segbot_simulator/segbot_gazebo

Multiple packages found with the same name "segbot_gui":
- segbot/segbot_gui
- segbot_apps/segbot_gui

Multiple packages found with the same name "segbot_logical_translator":
- segbot/segbot_logical_translator
- segbot_apps/segbot_logical_translator

Multiple packages found with the same name "segbot_navigation":
- segbot/segbot_navigation
- segbot_apps/segbot_navigation

Multiple packages found with the same name "segbot_simulation_apps":
- segbot/segbot_simulation_apps
- segbot_simulator/segbot_simulation_apps

shiqizhang6, 2015-07-30:
This error appears while running: bwi_launch/launch/simulation.launch
@piyushk

piyushk, 2015-07-30:
@shiqizhang6 We've merged a few repositories. Please run the following commands:
```
cd ~/catkin_ws/src
wstool rm segbot_apps segbot_simulator
rm -rf segbot_apps segbot_simulator
cd ~/catkin_ws
catkin_make clean
catkin_make
```

jack-oquin, 2015-07-30:
This is caused by us merging **segbot_simulator** and **segbot_apps** into **segbot**.

When you update your workspace, you need to delete the two other repositories.

jack-oquin, 2015-07-30:
We should send a warning note to the bwi mailing list.

piyushk, 2015-07-30:
Already done :+1:.

closed 2015-07-30 by piyushk