utexas-bwi/segbot #61: segbot_logical_translator: build farm failure resolving bwi_msgs/ResolveChangeFloor.h

jack-oquin, 2016-05-25:
```
Scanning dependencies of target segbot_logical_navigator
100%] Building CXX object CMakeFiles/segbot_logical_navigator.dir/src/nodes/segbot_logical_navigator.cpp.o
/tmp/catkin_workspace/src/segbot/segbot_logical_translator/src/nodes/segbot_logical_navigator.cpp:59:41: fatal error: bwi_msgs/ResolveChangeFloor.h: No such file or directory
 #include <bwi_msgs/ResolveChangeFloor.h>
                                         ^
compilation terminated.
```

Ref: http://build.ros.org/job/Idev__segbot__ubuntu_trusty_amd64/6/changes

jack-oquin, 2016-06-03:
There is a missing CMake dependency on the message generation.

jack-oquin, 2016-06-03:
There is more than one problem here. It still fails, probably because the released version of bwi_common is too old:

 * http://build.ros.org/job/Idev__segbot__ubuntu_trusty_amd64/9/changes

jack-oquin, 2016-06-03:
@piyushk: I suspect we can fix this by releasing a new version of ``bwi_common``. 

That probably does not depend on those Stanley repositories.

piyushk, 2016-06-03:
That sounds fine to me.

jack-oquin, 2016-06-04:
Before releasing bwi_common, we should fix these problems:

Test failures:
```
[100%] Built target run_tests
joq@iron:build$ catkin_test_results test_results/
bwi_launch/roslaunch-check_launch.xml: 1 tests, 0 errors, 1 failures
segbot_bringup/roslaunch-check_launch_segbot_v1.launch.xml: 1 tests, 0 errors, 1 failures
segbot_bringup/roslaunch-check_launch_segbot_v2.launch.xml: 1 tests, 0 errors, 1 failures
segbot_bringup/roslaunch-check_launch_segway_base.launch.xml: 1 tests, 0 errors, 1 failures
segway_description/roslaunch-check_launch.xml: 1 tests, 0 errors, 1 failures
segway_navigation_apps/roslaunch-check_launch.xml: 1 tests, 0 errors, 1 failures
stop_base/rostest-tests_stop_base_passthru.xml: 1 tests, 0 errors, 1 failures
stop_base/rostest-tests_stop_base_requests.xml: 1 tests, 0 errors, 1 failures
Summary: 84 tests, 0 errors, 8 failures
```

ROS lint failures:
```
$ make roslint
Scanning dependencies of target roslint_stop_base
src/stop_base/requester_node.py:72:23: W503 line break before binary operator
src/stop_base/controller_node.py:92:13: E265 block comment should start with '# '
src/stop_base/transitions.py:57:1: E266 too many leading '#' for block comment
src/stop_base/transitions.py:108:35: W503 line break before binary operator
src/stop_base/transitions.py:109:35: W503 line break before binary operator
make[3]: *** [roslint_stop_base] Error 1
make[2]: *** [bwi_common/stop_base/CMakeFiles/roslint_stop_base.dir/all] Error 2
make[1]: *** [bwi_common/bwi_logging/CMakeFiles/roslint.dir/rule] Error 2
make: *** [roslint] Error 2
```

jack-oquin, 2016-06-04:
fixed roslint errors in stop_base

closed 2018-03-02 by jack-oquin