utexas-bwi/segbot #62: segbot_sensors: SegbotVelodyneOutlierRemovalConfig.h missing dependency

shiqizhang6, 2016-05-27:
This is the printout error: 

/home/users/szhang/catkin_ws_scavenger/src/segbot/segbot_sensors/include/segbot_sensors/segbot_velodyne_outlier_removal.h:60:63:
fatal error: segbot_sensors/SegbotVelodyneOutlierRemovalConfig.h: No such file or directory
#include <segbot_sensors/SegbotVelodyneOutlierRemovalConfig.h>

shiqizhang6, 2016-05-27:
@piyushk @maxsvetlik

jack-oquin, 2016-05-27:
Probably a missing dependency on the gencfg output (or something like that).

Things like that might work on one machine and fail on another due to build parallelism and the number of processors available.

shiqizhang6, 2016-05-31:
@jack-oquin Do you mean the issue comes from our use of dynamic reconfiguration or you are pointing to a solution? Sorry that I am new to dynamic reconfiguration.

jack-oquin, 2016-05-31:
I was gone Sunday and Monday. I'll take a look at it today.

I am just guessing what the problem may be.

jack-oquin, 2016-05-31:
I tried to reproduce this problem in nibbler, but it built without error. I'll try another lab machine.

jack-oquin, 2016-06-03:
Still working for me. I'll try to find the problem tomorrow via code review.

shiqizhang6, 2016-06-03:
It builds well on the robots, but not on the lab machines. I can try again tomorrow on the lab machines and provide more detailed info. Thanks! @jack-oquin

jack-oquin, 2016-06-03:
I found the bug, fixing it now.

jack-oquin, 2016-06-03:
I am pretty sure c95eff2 fixes the problem.

Please re-open if I am wrong and it happens again. :smile:

closed 2016-06-03 by jack-oquin

shiqizhang6, 2016-06-03:
@jack-oquin Thanks! It compiles on the lab machines now.