utexas-bwi/segbot #68: segbot_launch: monitor computer temperatures, log to diagnostics

jack-oquin, 2016-06-28:
Something like https://github.com/ros-drivers/linux_peripheral_interfaces should work for this.

jack-oquin, 2016-06-28:
The segbot_v3 computer case has been observed getting too hot to touch comfortably.

jack-oquin, 2016-06-28:
See also: http://wiki.ros.org/libsensors_monitor

piyushk, 2016-06-29:
I'm manually monitoring the temperature on the robot that Shiqi's using. After one hour of high usage (full nav stack, person detection, virtour, velodyne, hokuyo, kinect), the core temperature is about 48C. I'll continue monitoring it over the next 2 hours. A high temperature is 79C, and 85C is crit. Somewhere in-between the CPU will get throttled. 

Please note that the entire case is a heat-sink, so it won't be unusual for it to get reasonably hot.

piyushk, 2016-06-29:
After 2 hours of continuous usage, the core temperature is at 65C, which is a pretty sharp increase from an hour ago.

jack-oquin, 2016-06-29:
Whether it is a problem with the current hardware or not, monitoring temperatures is good basic engineering.

piyushk, 2016-06-29:
I agree. 

The temp didn't really cross 65C at the end of three hours, but the case had become warmer. I'll think about this a bit more prior to the next round of construction.