utexas-bwi/segbot #83: Segbot_v3 Controller Not Working

sxtejeda, 2017-06-30:
After updating the workspace, running 
$roslaunch bwi_launch segbot_v3.launch
causes the segbot_v3_controller located in segbot/segbot_bringup/scripts to crash. It may be worth noting that non-updated accounts (e.g. the public fri account) boots up this nodelet without any issue.

jack-oquin, 2017-07-03:
Commit e8505b48ba695914ccfe960a4fda160936e10b9f should have fixed this. Please update and try again.

sxtejeda, 2017-07-03:
I updated it, and the launch is still not working. The current error is

"Platform ID is not correct!!! Platform reports RMP_210, user set RMP_440..."

I am not sure if the actual segway base is RMP_210 or RMP_440, but I believe that the user specification is coming from segbot_v3.launch, line 58:
```
<node pkg="segway_ros" ns="segway" type="segway_driver" name="segway_driver" respawn="true" output="screen">
  <param name="platform" value="$(optenv SEGWAY_BASE_PLATFORM false)" />
</node>
```

where echoing SEGWAY_BASE_PLATFORM on the v3 returns RMP_440

sxtejeda, 2017-07-03:
Update: changing SEGWAY_BASE_PLATFORM to RMP_210 fixes the problem, but the only way I know how to do this is just through the export command. Does someone know where this variable is originally defined?

jack-oquin, 2017-07-03:
It's supposed to be defined in your ``~/.bashrc``:
```
## Segbot V3 settings
export ROBOT_NETWORK=wlan4
export SEGWAY_BASE_PLATFORM=RMP_210
export SEGWAY_HAS_BSA=false
export SEGWAY_HAS_ONBOARD_JOY=true
export SEGWAY_PLATFORM_NAME=RMP_210
export SEGWAY_POWERS_PC_ONBOARD=true
export SEGWAY_RUNS_IN_BALANCE_MODE=false
```

I doubt all of that is actually necessary, but that's what I'm currently using.

That needs to be clearly documented somewhere prominent.

jack-oquin, 2017-07-04:
Sadly, the segway_v3 software from Stanley Innovations requires a large number of environment variable to configure the correct robot.

If someone could figure out a clean way to define these in a system-wide location, they would probably accept a pull request to remove that messy interface.

sxtejeda, 2017-07-06:
Update: I have found where $SEGWAY_BASE_PLATFORM is being defined! It's in segway_v3/segway_v3_config/segway_config.bash

I would alter the export, but this config file is also changing $SEGWAY_440LE_WHEEL_TRACK as well as $SEGWAY_440LE_WHEEL_OFFSET due to it it being defined as, specifically, an RMP_440_LE_ECI. Should I just open up a new issue over there?

jack-oquin, 2017-07-14:
That just sets the default configuration for all of their platforms. They're not going to want to change that.

What we need is a way to override it in the local environment. The ``~/.bashrc`` works, but is awkward. Setting it in a per-system place like ``/etc/environment`` on each robot would be good.

sxtejeda, 2017-07-14:
Placing it in etc won't work - segway_config.bash is loaded in .bashrc  whenever the workspace is sourced, so the new export has to happen after that action. Environment is loaded far before that, so it ends up being overwritten. If there is some way to protect an environmental variable from being overwritten, then we could place it in etc, but otherwise the only real fix is to define it after sourcing the workspace.

jack-oquin, 2017-07-14:
By the way, our actual robots are RMP_110, and not 210's. 

Formerly, they had no explicit support for the 110. Now, the latest Stanley version *does* support 110, and requires setting the base platform variable to that. I tested that Wednesday on the setbot_v4, roberto. I think the segbot_v3, leela, may work differently.  See:  StanleyInnovation/segway_v3#26

jack-oquin, 2017-07-14:
I am confused about when sourcing those variables takes effect. 

What you say makes sense, but then it should also override the ``.bashrc`` settings, so maybe it's done conditionally somehow.

sxtejeda, 2017-07-14:
From what I understood, the file where the Segway base is initially defined is done so once the workspace is sourced.  I think it's just done sequentially, because if you export the variable before sourcing the workspace, or if you export it in etc, the variable is still defined as RMP_440.

jack-oquin, 2017-07-14:
I *never* source the workspace in my .bashrc. That's because I often use more than one.

So, there is something else going on.

jack-oquin, 2017-09-12:
There does not appear to be any further work associated with this issue, so I'm closing it now.

closed 2017-09-12 by jack-oquin