utexas-bwi/segbot #98: Travis CI: fix failures

jack-oquin, 2018-01-31:
The solution will probably be similar to what was done for utexas-bwi/bwi_common#102

jack-oquin, 2018-01-31:
The Kinetic build failed with the following error:
```
No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
The build has been terminated
```

jack-oquin, 2018-01-31:
Not sure what the cause of the timeout problem is. There are an awful lot of packages being built for Kinetic. It would be nice if more of the dependencies could be released as binaries.

jack-oquin, 2018-01-31:
The fact that the Indigo tests are working is encouraging. They require fewer source repositories to build.

jack-oquin, 2018-02-01:
I attempted a suggested work-around for Travis jobs timing out, but it did not work with the ROS-I tools.

see: https://docs.travis-ci.com/user/common-build-problems/#My-builds-are-timing-out

jack-oquin, 2018-02-02:
I have not made any progress in solving the Travis timeout with the Kinetic branch. Things I've tried:

 * [Using travis_wait to increase the timeout](https://docs.travis-ci.com/user/common-build-problems/#My-builds-are-timing-out)
 * [Running the CI job locally](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#run-industrial-ci-on-local-host) (both ways)

130s, 2018-02-02:
I know this is a dumb thing to ask, but do builds on 1) local computer 2) if possible on Docker (since with industrial_ci each job runs on Docker) pass and/or leave any hints?

If you're curious if this is industrial_ci related I recommend opening a ticket there (I'm a maintainer. We'll like to hear users' issues).

jack-oquin, 2018-02-02:
Isaac! Good to hear from you. I really admire your excellent work on **industrial_ci**.

I have no reason to believe our current problems are in **industrial_ci**. I did run tests on my local computer using both **run_ci** and **run_travis** (I believe those are with and without docker), but so far have not succeeded. I no longer have the shell output from those attempts. In one case, package dependencies included within the workspace were not getting resolved by rosdep. I probably did not configure my workspace correctly for that, somehow.

I suspect that the root cause of our difficulty on Kinetic is too many repositories and packages being built in a single Travis job, because we are still building several repo dependencies that have not been released to ROS Kinetic. Our Indigo branch tests are all working fine.

Also, one of our dependencies is a CMake library package, [libsegwayrmp](https://github.com/utexas-bwi/libsegwayrmp), which is not catkin, although it does have a ``package.xml``. We build it from source using catkin tools, but maybe it's a problem for **industrial_ci**? Your docs say only catkin packages are supported.

I am thinking of ways to break the tests up into more manageable chunks, maybe creating our own PPA for some packages we can't release into the ROS repositories. That would be worthwhile, even if some other problem turns out to be at work here.

jack-oquin, 2018-02-02:
Just building our whole code base on Kinetic is badly bogging down one of the old segbot_v2 laptops.

We need to create methods like using catkin tools whitelists that make it easy to compile only a useful subset of the packages.

jack-oquin, 2018-02-02:
It took a little over an hour to build all the Kinetic repo packages on **bender**, which has an old Dell laptop.

jack-oquin, 2018-02-03:
It turns out catkin build can use a metapackage as a whitelist, building only its dependencies (recursively). That way we can check out the whole repository, but only build segbot_v2 packages, or similarly for the other platforms.

jack-oquin, 2018-02-03:
I divided the tests into two parts: **only_segbot_arm** and **only_segbot_base**. The segbot base test runs successfully, but the segbot arm test fails with a similar timeout after starting **segbot_arm_manipulation**.

See: https://travis-ci.org/utexas-bwi/segbot/jobs/337009286

jack-oquin, 2018-02-03:
Commit e5d0a38 is an attempt to get the kinetic branch test to succeed by commenting out the segbot arm tests.

jack-oquin, 2018-02-03:
That worked, but we still need to figure out what's wrong with the commented-out arm tests.

jack-oquin, 2018-02-05:
@nickswalker: I'm going to leave the arm tests commented out for now. At least, the other packages can now get some CI testing. 

If you want to experiment with the arm tests, uncomment the [three lines here](https://github.com/utexas-bwi/segbot/blob/kinetic/.travis.yml#L15), and push to some experiemental branch. I've been using **travis-kinetic** for that, deleting it to avoid confusion, as it tends to diverge from the real **kinetic** branch.

Meanwhile, I'm going to work on creating a custom BWI package repository, so we don't normally need to build things like the segway drivers, kinova-ros, or agile_grasp. That may end up solving this issue, too.

jack-oquin, 2018-02-06:
That last commit was supposed to run the arm tests, but not fail the whole job if they did not succeed. That didn't work.

jack-oquin, 2018-02-06:
The original Travis config no longer works for Kinetic.

closed 2018-03-02 by jack-oquin