utexas-bwi/segbot #99: Kinetic: navigation getting stuck

jack-oquin, 2018-02-14:
@nickswalker observed this on bender (segbot_v2). He says that it looks like ghost obstacles are appearing in the laser scan. 

Nick was running **visit_door_list** and the robot got stuck at an angle near the trash cans. There was a bogus laser scan point keeping it from going the opposite direction. The shell output was full of yellow warning messages.

There is no evidence yet, that this problem is unique to that specific particular robot model.

All the robots have been observed weaving "drunkenly" navigating on Kinetic. It is possible that ghost points could cause that behavior, too.

jack-oquin, 2018-02-14:
@harely22: Roberto got stuck on the corner of one of the bookshelves near the wide spot at the East end of the 3N lab area. I forgot to check whether you are running Kinetic on it. I didn't see any ghost obstacles in the rviz display. I backed the robot up a short distance using **teleop_keyboard**, and it started running again.

I was running Kinetic on pickles at the same time, but could not tell if the problem was caused by interactions between those two robots.

harely22, 2018-02-14:
I noticed that in a couple of times the robot  (Roberto in both cases) got stuck next to a pod, physically touching the pod separation. The localization in both times showed that it thought that part of it was inside the wall. To release it I gave it a navigation to a close goal location and it continued visiting rooms from there. As Jack suggested, this could be used as a possible solution to such cases. BTW, there were no other robots running simultaneously.

jack-oquin, 2018-02-15:
Sounds like the roberto problem is (mostly) a separate issue. Maybe we should open one for that specific platform. It may be due to the size of the wheels or some other platform difference.

jack-oquin, 2018-03-02:
@FernandezR discovered some parameter bugs (#107) in the Hokuyo driver configuration that may explain some or even most of these problems.