utexas-bwi/segbot #107: Kinetic: Hokuyo URG Node Param Names Incorrect

FernandezR, 2018-03-01:
max_ang and min_ang params should be angle_max and angle_min for the urg node in kinetic. This could be causing some of our laser errors on the robots.

jack-oquin, 2018-03-01:
Good. Would you commit those fixes to the "kinetic" branch?

FernandezR, 2018-03-02:
Yes, I'll take care of it.

FernandezR, 2018-03-02:
@jack-oquin Since you removed the angle filter hokuyo_filters_v2.yaml I believe we should add the angle_min and angle_max params from the hokuyo.launch file to the ethernet hokuyo launch file.

FernandezR, 2018-03-02:
I also believe we should be using the default param "scan_filtered" for localization_scan_topic in segbot.launch.xml for amcl on the v4 instead of the current param "scan" that is passed by  segbot_v4.launch.

jack-oquin, 2018-03-02:
You are probably right. Let's experiment with some of those parameters on the robots this afternoon. 

I think we could run the ``angle_range_filter`` now. It was commented out due to early problems building ``segbot_sensors_filters``. That seems to be working now, but the config file was never restored.

Thanks for investigating!

jack-oquin, 2018-03-02:
I ran with ``angle_range_filter`` active, and the results were obviously better. Navigation was reasonably smooth.

jack-oquin, 2018-03-02:
It seems to work fine with the angle min and max driver parameters: 4076825

closed 2020-01-31 by maxsvetlik