utexas-bwi/segbot #108: indigo branch depends on changes only present in bwi_common fern/led_study branch

rdelfin, 2018-03-09:
In the Indigo branch, file `segbot_led/src/led_control_server.cpp` includes `bwi_msgs/LEDCameraStatus.h`, which is only present in `bwi_common`'s `fern/led_study`. This breaks a clean clone of a new bwi clean install. Either `fern/led_study` needs to be merged into indigo or the changes that caused this dependency need to be moved out into a separate branch or repo.

I have not tested this issue in the `kinetic` branch, so I am unsure if the changes are there too.
@FernandezR

rdelfin, 2018-03-09:
I tried working on `fern/led_study` on both `bwi_common` and `segbot`, but a separate issue shows up with  `bwi_services/SpeakMessage.h`. As a temporary solution I went back to the `indigo` branches and deleted `segbot_led` locally. It seems to have fixed the issue, but we need a better solution.

FernandezR, 2018-03-09:
The main branch of indigo should be updated though it seems that it is not. I'll try to do a pull request with the changes.

FernandezR, 2018-03-09:
You should be fine now with the indigo branch, I made pull request and merged it.

closed 2018-03-09 by FernandezR